marvinroger/async-mqtt-client #226: Heap goes down when receiving messages with qos >0

luebbe, 2021-02-10:
tested with develop FullyFeatured-ESP8266 example. Added a `Serial.println(ESP.getFreeHeap());` to the end of `onMqttMessage(..)`
mosquitto on raspberry
MQTT Explorer on windows

Independend of the message size, the reported FreeHeap goes down with every received message
QoS 2: 40-200 Bytes
QoS 1: 20-60 Bytes
QoS 0: 16 Bytes for the first message, 0 Bytes for following QoS 0 messages if you send "rapidly", e.g. 1/second. 16 Bytes again, when you wait a few seconds.

bertmelis, 2021-02-10:
Didn't see this but I'm not arguing; there has been a lot of changes.

I'm very scarce of free time the next week so I can't do deep bug hunting.

Pablo2048, 2021-02-11:
I can confirm this - Testing with QoS 1 messages. Latest version without this heap decrease is https://github.com/marvinroger/async-mqtt-client/commit/97d681fa200e414b8d45297a10e8afe6f382068d ...

bertmelis, 2021-02-11:
> I can confirm this - Testing with QoS 1 messages. Latest version without this heap decrease is [97d681f](https://github.com/marvinroger/async-mqtt-client/commit/97d681fa200e414b8d45297a10e8afe6f382068d) ...

I changed something that wasn't meant to be changed when doing TLS. I can't reproduce now (I did very limited testing though).

Pablo2048, 2021-02-11:
Great. Seems to be fixed (at least for me...). Thanks.

luebbe, 2021-02-11:
Thanks @bertmelis. Will test again this evening

luebbe, 2021-02-12:
Fixed for me.

commit 2635abd0 (2021-02-11): fix cleanup

fixes #226
closed 2021-02-12 by luebbe