marvinroger/async-mqtt-client #229: WPS + mqtt does not work properly.

jhbirdchoi, 2021-02-19:
If you use ssid + password, mqtt works normally.
However, using WPS, mqtt does not work.

/*
Example Code To Get ESP32 To Connect To A Router Using WPS
===========================================================
This example code provides both Push Button method and Pin
based WPS entry to get your ESP connected to your WiFi router.

Hardware Requirements
========================
ESP32 and a Router having atleast one WPS functionality

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/
#include <AsyncMqttClient.h>
#include "WiFi.h"
#include "esp_wps.h"
#include "Arduino.h"
//#include "esp_camera.h"
//#include "esp_http_server.h"
//#include "esp_timer.h"
//#include "fb_gfx.h"
//#include "img_converters.h"
//#include "soc/soc.h" //disable brownout problems
//#include "soc/rtc_cntl_reg.h"  //disable brownout problems

// Raspberry Pi Mosquitto MQTT Broker
#define MQTT_HOST "mqtt-dashboard.com"
#define MQTT_PORT 1883// Raspberry Pi Mosquitto MQTT Broker

AsyncMqttClient mqttClient;
/*
Change the definition of the WPS mode
from WPS_TYPE_PBC to WPS_TYPE_PIN in
the case that you are using pin type
WPS
*/
#define ESP_WPS_MODE      WPS_TYPE_PBC
#define ESP_MANUFACTURER  "ESPRESSIF"
#define ESP_MODEL_NUMBER  "ESP32"
#define ESP_MODEL_NAME    "ESPRESSIF IOT"
#define ESP_DEVICE_NAME   "ESP STATION"

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
  mqttClient.publish("check", 2, true, String("aaa").c_str());
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");
  if (WiFi.isConnected()) {
    //ToDo
  }
}

void onMqttPublish(uint16_t packetId) {
  Serial.print("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

static esp_wps_config_t wps_config;

void wpsInitConfig(){
  wps_config.crypto_funcs = &g_wifi_default_wps_crypto_funcs;
  wps_config.wps_type = ESP_WPS_MODE;
  strcpy(wps_config.factory_info.manufacturer, ESP_MANUFACTURER);
  strcpy(wps_config.factory_info.model_number, ESP_MODEL_NUMBER);
  strcpy(wps_config.factory_info.model_name, ESP_MODEL_NAME);
  strcpy(wps_config.factory_info.device_name, ESP_DEVICE_NAME);
}

String wpspin2string(uint8_t a[]){
  char wps_pin[9];
  for(int i=0;i<8;i++){
    wps_pin[i] = a[i];
  }
  wps_pin[8] = '\0';
  return (String)wps_pin;
}

void WiFiEvent(WiFiEvent_t event, system_event_info_t info){
  switch(event){
    case SYSTEM_EVENT_STA_START:
      Serial.println("Station Mode Started");
      break;
    case SYSTEM_EVENT_STA_GOT_IP:
      Serial.println("Connected to :" + String(WiFi.SSID()));
      Serial.print("Got IP: ");
      Serial.println(WiFi.localIP());
      connectToMqtt();
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("Disconnected from station, attempting reconnection");
      connectToMqtt();
      WiFi.reconnect();
      break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
      Serial.println("WPS Successfull, stopping WPS and connecting to: " + String(WiFi.SSID()));
      esp_wifi_wps_disable();
      delay(10);
      WiFi.begin();
      break;
    case SYSTEM_EVENT_STA_WPS_ER_FAILED:
      Serial.println("WPS Failed, retrying");
      esp_wifi_wps_disable();
      esp_wifi_wps_enable(&wps_config);
      esp_wifi_wps_start(0);
      break;
    case SYSTEM_EVENT_STA_WPS_ER_TIMEOUT:
      Serial.println("WPS Timedout, retrying");
      esp_wifi_wps_disable();
      esp_wifi_wps_enable(&wps_config);
      esp_wifi_wps_start(0);
      break;
    case SYSTEM_EVENT_STA_WPS_ER_PIN:
      Serial.println("WPS_PIN = " + wpspin2string(info.sta_er_pin.pin_code));
      break;
    default:
      break;
  }
}

void setup(){
   
  Serial.begin(115200);
  delay(10);

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif
  
  Serial.println();

  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_MODE_STA);

  Serial.println("Starting WPS");
  
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);
  
  wpsInitConfig();
  esp_wifi_wps_enable(&wps_config);
  esp_wifi_wps_start(0);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  Serial.printf("before sned..mqtt conncted = %d\n", mqttClient.connected());
  mqttClient.publish("check", 2, true, String("aaa").c_str());
  Serial.println("-send mqtt...");
}

void loop(){
  //nothing to do here
}

--consoe log---
Starting WPS
Station Mode Started
...................................................................................................................................................................................................WPS Successfull, stopping WPS and connecting to: U+NetBFD4
Disconnected from station, attempting reconnection
Connecting to MQTT...
.........Connected to :U+NetBFD4
Got IP: 192.168.123.110
Connecting to MQTT...
.before sned..mqtt conncted = 0
-send mqtt...
Connected to MQTT.
Session present: 0

Is there anything I missed?

thanks

bertmelis, 2021-02-19:
It says "connected to mqtt". What did you expect?

Make sure you use the develop branch of the lib.

bertmelis, 2021-02-20:
Mind that this lib in asynchronous. Functions return immediately and do not block or wait for the desired outcome.

You publish immediately after calling connect. But the connection is not ready by then. Try to publish in the onConnected function.

closed 2021-02-20 by bertmelis