marvinroger/async-mqtt-client #238: Publish when not connected

bertmelis, 2021-03-02:
Should it be made possible to publish with qos > 0 when not connected?

Pablo2048, 2021-03-02:
Maybe with help of LittleFS/any another persistent storage IMO. But then the library will be huge and over complicated. Also I don't think that storing all the messages with QoS > 0 to persistent storage is the best, what we can do to the Flash memory and/or filesystem.

luebbe, 2021-03-02:
I don't think, that this is really necessary. You would have to keep a dictionary of messages and overwrite old messages when a new message is sent to the same topic. What happens if you want to send every second (or faster). Do you want to implement FS caching in order not to burn out the flash memory?

bertmelis, 2021-03-02:
I don't want to do persistent storage. I keep this example from the MQTT spec in mind: http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718097

Anyone capable of building a parking payment terminal should also be capable of altering this lib :smile: 

But just to make sure: I don't have any preference. I was just wondering. The spec isn't very clear on this.

proddy, 2021-03-02:
I don't think you should publish either. In my project I store all MQTT messages in a queue (both publish and subscribe commands) and the queue is processed on a timer every 200ms. This was to overcome flooding on slow networks on the ESP8266. Then I implemented a holding-area for qos1 & 2 (basically if there was no PUBACK re-send the MQTT message again, up to 3 attempts). It was a shit load of code and I don't want to do it again, and wouldn't recommend it, as it kills the heap memory.

If you want to cache on qos>0 then maybe store the last 2 messages only, in a circular buffer.

Pablo2048, 2021-03-02:
Ok, so lets just think:
  * should the unsent messages survive PoR?
  * if yes -> persistent storage (and I don't suggest to go this way)
  * if no - proddy's solution is imo the best - define small queue/memory area for storing the unsent messages and inform caller about not enough space for storing, or kill the oldest one (not very nice...)
Anyway I don't think that this should be handled with the library. Maybe some another QueueHelper class?

bertmelis, 2021-03-05:
closing. publishing when not connected will not be enabled.

closed 2021-03-05 by bertmelis