marvinroger/async-mqtt-client #240: Allow supplying a configured Client object?

Duckle29, 2021-03-28:
Hi. I've been working on a device that has to use external MQTTS services, and as such it's a must to have proper TLS capabilities, not based on a quickly expiring fingerprint, but instead a proper root certificate store. BearSSL supports this, and it's working great in my current testing with a different library. So far I've gotten it mostly working with Adafruits mqtt library, however ran into some deal-breaking bugs that seem to have been known since 2017. As such I kept looking and found this.

Basically I have a `WiFiClientSecure` object, which is an inherited class all the way down to `Client`. This means it works with anything that works with such clients, but from what I can tell, AsyncClient isn't related to any such Client class. I'm guessing that's because the normal Client can't be Async?
If that is indeed the case, am I right in assuming it won't be possible to have async-mqtt-client work with a `Client` object, as it wouldn't be async anymore? 

I basically want to ask someone more familiar with the codebase, if what I have in mind is a huge task or not. I'm leaning towards it is :/

For context, my current implementation is here: https://github.com/Duckle29/wifir_firmware/tree/main/lib/ssl_wrapper

bertmelis, 2021-03-28:
Your fears are true, it just isn't compatible.

Duckle29, 2021-03-28:
Dang. Oh well there are some fixes floating around for the bug. Guess I'll be working off of yet another github lib_deps :/

Thanks for quick response, and I wish I could use your lib. It looks dope! :)

closed 2021-03-28 by Duckle29