marvinroger/async-mqtt-client #241: Async mqtt client unusable if ping is filtered

Vfgandara, 2021-03-29:
Hello, for some reason my ISP is blocking ping requests, i tried contacting them and all but it seems i can't do much about it. This renders this wonderful library unusable since then i can't connect to the mqtt server (it seems that pinging is necessary to do so). Is there any alternative to ping to ensure connectivity? Although there is this ping request blocking from my ISP, everything else works perfectly, eg I can connect via SSH to said server and all, it's literally only the ping request that's not working properly.

bertmelis, 2021-03-29:
AsyncTCP and this lib doesn't need ping.
Are you connecting via hostname or via IP?
Are you using TLS?

Vfgandara, 2021-03-29:
Dear @bertmelis,

Thanks for your prompt reply. I am not using TLS. Connecting using IP or hostname. However, notice that ping request is used on every onPoll request. See here: 
https://github.com/marvinroger/async-mqtt-client/blob/16e5ede4e3565c0374e71e32874ae6387d69e7fc/src/AsyncMqttClient.cpp#L461-L487

Pablo2048, 2021-03-29:
Definitely not - you are mixing two things. Your mentioned code is MQTT ping - some sort of keepalive signaling ( http://www.steves-internet-guide.com/mqtt-keep-alive-by-example/ ), but ping requests from the computer are the ICMP protocol...

Pablo2048, 2021-06-08:
I think that this can be closed now...

closed 2021-06-08 by luebbe