marvinroger/async-mqtt-client #242: Messages not arriving after ESP32 deep sleep

goofy2k, 2021-05-09:
Hi,

I have a battery operated ESP32 board (=client) and use deep sleep mode to save power.  I want to receive messages that have been sent by Nodered to the broker while the board (client) was asleep.

After a fresh reboot, messages are received properly. But when the board has awoken from it first sleep, the messages don't arrive. I added  **mqttClient.setCleanSession(false);**  to my initialisation code.   This does not help.

I subscribe to the topic everytime (after a first boot or after waking up).  When I only subscribe to the topic after  **if (sessionPresent == 0)**  the problem remains.

Does anyone have suggestions how to move on???

bertmelis, 2021-05-11:
You don't have to resubscribe. Subscriptions are part of session data.

Do you disconnect before going to sleep? Which QoS level do you use to send the messages?

goofy2k, 2021-05-11:
Thanks for you reply Bert. 

I do not explicitly disconnect (WiFi nor MQTT). BUT, I run the same code after a boot or after a wake up.  Maybe the connection procedure makes that it is not considered as the same session by the broker?  I have an option to bypass the connection procedure when it is a wake up........

 I tried to use all three QoS levels  for sending.

bertmelis, 2021-05-12:
You have to reconnect with the same client id and with clean session to false. On reconnect, does the broker return the "session present" flag?

Off course, only QoS>0 will be held for delivery.

goofy2k, 2021-05-12:
Yeah,  I believe my code meets all of the above requirements.  In the meantime I have a (more or less) working code based on PubSubClient.  I want to bring that to a certain level, before I am going to retry async-mqtt-client.  The asynchronous character is an important advantage. 

In general, I believe that reliability of the (re)connection is a key success factor.

bertmelis, 2021-05-13:
What is you keepalive time?
Because you don't disconnect, the broker keeps the connection open (half-open) and only closes the connection when the keepalive kick in.

Are you using the `develop` branch of this lib? If so, you can send everything and immediately disconnect (not force). The lib will wait until all outgoing messages are processed.

goofy2k, 2021-05-13:
I have to learn...  I did not assign a value to the keep alive consciously. I suppose (pls correct me if I am wrong) that the connection is closed automatically if there is no activity between client and broker during the keep alive period since the last activity. What would be a good strategy if I use the non-dev branch (master?).  ?   If I am correct, we have the following options:

1. Send everything and disconnect actively (client.disconnect()?) before going to sleep.
2. Send and do not disconnect actively
3. Send and disconnect by setting keepalive shorter then the expected time to go asleep.
4. Send and do not disconnect at all, by choosing a large keep alive.

I do not use the dev branch. So you suggest to send everything, and disconnect immediately. But I do not understand what you mean by disconnnect (not force).   I'll read the docs with the dev branch. I guess that because of the feature that outgoing messages will be processed even if you call disconnect or if the keep alive passed, there is also a command for forced disconnect.

Hopefully you want to make a few remarks yo help me understand the options a bit better.

Of course the ultimate question is what is the best option to have 

- reliable receipt of QoS > 0 messages
- reliable wake up cycles (prevent skipping of cycles because of slow connection or missed connection after wake up).

bertmelis, 2021-05-13:
The difference between the master branch and the develop branch is that in develop all outgoing messages are queued. So you can send everything you want and immediately disconnect. The disconnect will wait until the messages are published.

I'd suggest you use the develop branch and disconnect before sleep.

The commands did not change between master and develop. So you can safely read through the docs in master (except for the points about memory usage).

Keepalive can be kept as is (if not set, it takes the default 15sec).

I am not able to test myself at the moment. Might have some spare time next week.

goofy2k, 2021-05-13:
Thanks Bert,

I will do the tests myself. As I branched my project also (PubSubClient vs async_mqtt_client) it will take me some days. I'll keep you informed about the progress.

One question: what will happen if my app requires more connection time than the 15 seconds (keep alive) ?

bertmelis, 2021-05-13:
Keepalive is a mechanism to prevent half-open connections. When you are not publishing (or receiving) the ESP will sent small packets at regular intervals to keep the connection alive.

When the broker doesn't respond, the ESP will close the connection. Also, if the broker doesn't receive the packets in time, the broker will close the connection.

If you have a sensor that publishes every minute and the keepalive is 2minutes, no keepalive packet will be sent. If your keepalive is 1 minute and you only send every 2 minutes, an intermediate keepalive is sent to tell the broker not to disconnect.

A too long keepalive causes the broker or client to notice connection problems too late. When too short you will generate unnecessary traffic.

Sending the keepalive happens in the background.

goofy2k, 2021-05-16:
I have a basic app up and running !  I have used the master branch and default keepalive.  My previous attempts also contained some functionality (readout of battery voltage with the on-board ADC) which may have disturbed timing.   I can now send MQTT messages when the boards are asleep. They receive the messages OK after waking up!   I can now proceed with the functional payload of the app.  Thanks for your support.

I'll keep the recommendations in mind .....

closed 2021-05-19 by bertmelis