marvinroger/async-mqtt-client #243: CONN_ACK packet return code needed!

dudanov, 2021-05-21:
Hello everyone.

In my case, I need return code in the connection acknowledge packet.

It does not return to either the ``onConnect`` function or the ``onDisconnect`` function.
As a result, for example, with an authorization error code 5 (not authorized), no function is called, only ``onDisconnect`` and only when the TCP connection is disconnect.

It seems that the ``onDisconnect`` function provides the corresponding codes, but as I understand it, they are still not used?

In general, in my opinion, the ``onConnect`` function should be called with a connection status argument, not ``onDisconnect``. Yes, adding an argument to the callback function will break existing dependent custom code, but this functionality needs to be implemented.

Thanks.

bertmelis, 2021-05-21:
The client was written for the 3.1.1 version of the MQTT spec and this is a "SHOULD" instead of "MUST". I guess that it is why it was never implemented.

Should not take too much work... PRs are welcome. I might find some time in the weekend.

bertmelis, 2021-05-21:
Question: why would you want to have it in the onConnect callback and not in the onDisconnect?

1. yes, it is not backwards compatible
2. when the connection is refused, the broker MUST close the connection, so the onDisconnect callback is called.
3. if onconnect is called, you know for sure the return code is zero. Hence, you're only interested in the return code when the connection is refused.

dudanov, 2021-05-21:
> Question: why would you want to have it in the onConnect callback and not in the onDisconnect?

This is the packet returned in response to a connection request. Therefore, it would be more logical to return the status code to ``onConnect``, and not ``onDisconnect``. But it doesn't matter to me. The main thing is to solve the problem of recognizing the cause of the connection failure.

A bit of specifics. In my case, authentication on the server is by token. If the server administrator deletes the device record, then it must recognize the authentication error and start the automatic registration process according to a certain algorithm. And now I just cannot understand what is the reason for the disconnection, it is called ``onDisconnect`` with the reason ``AsyncMqttClientDisconnectReason::TCP_DISCONNECTED``, while the reason is an authentication error.

dudanov, 2021-05-21:
> Should not take too much work... PRs are welcome. I might find some time in the weekend.

In principle, I can also make changes and open the PR. How we do it? We look at the return code in response to the connection. If the code is nonzero, then we call the callback ``onDisconnect`` with this code and ignore the future callback from the ``AsyncTCP`` or let it be called again with ``AsyncMqttClientDisconnectReason::TCP_DISCONNECTED`` reason?

bertmelis, 2021-05-21:
I think I already got it. Do you mind testing the `develop` branch?
I'll also implement it in `master` if you think it's ok.

dudanov, 2021-05-21:
> I think I already got it. Do you mind testing the `develop` branch?

No, I didn't use the development branch.

> I'll also implement it in `master` if you think it's ok.

Yes, that's what you need! You definitely need to merge this changes with ``master`` branch and close this issue. Thank you!

bertmelis, 2021-05-21:
Did you test it? Or just reviewed the code?
I don't think much can go wrong with the small change... But a second pair of eyes are invaluable.

dudanov, 2021-05-24:
> Did you test it? Or just reviewed the code?

Now I checked it at work. The patch works fine. Thanks.

bertmelis, 2021-05-24:
Thank you, I'll merge to master soon (after which I'll close this issue).

> But in general, problems arose in the work of the library. Perhaps because of the development branch. I do not know.

I'm happy to hear from your problems as I don't seem to have any.

dudanov, 2021-05-24:
> Thank you, I'll merge to master soon (after which I'll close this issue).
> 
> > But in general, problems arose in the work of the library. Perhaps because of the development branch. I do not know.
> 
> I'm happy to hear from your problems as I don't seem to have any.

Oops!.. Sorry. I'm open PR too. #245

dudanov, 2021-05-24:
> I'm happy to hear from your problems as I don't seem to have any.

I'm using ``https://github.com/bertmelis/async-mqtt-client`` develop branch.
The library has ceased to function normally. I had to create a fork, make changes myself, test and create a PR.

dudanov, 2021-05-24:
@bertmelis Look at #244 . You forgot about ``ESP8266_NOT_ENOUGH_SPACE`` reason.

bertmelis, 2021-05-24:
> @bertmelis Look at #244 . You forgot about `ESP8266_NOT_ENOUGH_SPACE` reason.

100% right.

Merged into `master`. I'll update the `develop` branch too.

closed 2021-05-24 by bertmelis