marvinroger/async-mqtt-client #247: Callback hangs

goofy2k, 2021-06-13:
Hi, I have a problem that my app hangs after calling the On Message callback.  Is there e.g. a max length of the message payload?   I will also try a new callback that does not contain my own code. And I willl try the original callback with a short payload...

Thanks
Fred

proddy, 2021-06-13:
stating the obvious here just in case: make sure there are no blocking delay()'s in your callback function

goofy2k, 2021-06-13:
Thanks! I am going to test some variants....

goofy2k, 2021-06-13:
I did some testing. I did some tricky handling of the payload and topic data in the callback . It made that the callback never returned. 

I am struggling again and again with null terminated strings and byte buffers..... and their lengths.
I had the idea of storing the incoming payloads in a stack with the objective of handling them later. I will abandon this idea. Too complicated for me ....

bertmelis, 2021-06-13:
> I did some testing. I did some tricky handling of the payload and topic data in the callback . It made that the callback never returned.
>
> I am struggling again and again with null terminated strings [...]

It shouldn't be that difficult, give me a moment

goofy2k, 2021-06-13:
Oops, I like your enthousiam! There are other reasons for my current
approach... So, don't spend a lot of time on it Bert :-).

One other reason is:

I created an API for my boards. The commands come from a (local) cloud
server, which is the manager for a fleet of microcontrollers. Amongst
others the cloud server can trigger firmware updates by OTA. The commands
and parameters (such as filename and path for new firmware) are sent by
MQTT. Of course I dont want to start an update in the onMQTTmessage
callback. In the callback I set some flags (e.g. updateavail) and
parameters that I store for later use. Updates are carried out after
finishing the normal tasks of the boards,  just before going to sleep.

I probably also want to persist those flags in SPIFFS, such that tasks
could be carried out in (at the very beginning or the end) a next awake
cycle. It is probably counter productive (space wise) to save complete
message payloads.

Nevertheless:  I could definitely learn from your proposal!

Fred

goofy2k, 2021-06-13:
And yes: when I send more than one message to the boards while they are
asleep I see the incoming messages appear in a row, nicely as if they come
from a stack.

In fact the stack is maintained by the MQTT broker, which is good! I want
the boards to focus on their own tasks....:-)

bertmelis, 2021-06-14:
Ok.

I'll close this issue as it is probably an error in your user code. Feel free to reopen or just continue the discussion if needed.

closed 2021-06-14 by bertmelis