marvinroger/async-mqtt-client #257: Async MQTT client parallel to ASYNC WifiManager and Async Webserver

miloit, 2021-07-30:
Hello,

I want to use Async MQTT client parallel to ASYNC WifiManager and Async Webserver

here is my code

```
#include <LittleFS.h>                   //this needs to be first, or it all crashes and burns...

#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>
#endif

//needed for library
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>         //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <PubSubClient.h>
#include <ESP8266mDNS.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>
#include <Arduino.h>

#define NUMBER_OF_SUBCRIBINGTOPICS 3

//define your default values here, if there are different values in config.json, they are overwritten.
char mqttServer[40];
char mqttPort[40];
char mqttUser[40];
char mqttPassword[40];
char mqttFingerprint[80];
char deviceName[40];
char softwareVersion[] = "0.1";
char mqttSubscribeTopics[NUMBER_OF_SUBCRIBINGTOPICS][150] =
{ "/text",
  "/intensity",
  "/scrollwait"
};

const char* PARAM_MESSAGE = "message";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

AsyncWebServer server(80);
DNSServer dns;
WiFiClientSecure espWifiClient;
PubSubClient client(espWifiClient);
MDNSResponder mdns;
AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;
WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;
AsyncWiFiManager wifiManager(&server,&dns);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

[... 457 lines cut ...]
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}
```
But I always get
```
Disconnected from MQTT.
Connecting to MQTT...
Disconnected from MQTT.
```

Any help appriciated

luebbe, 2021-07-30:
Maybe you should first try to build it up step by step. Remove the unnecessary stuff, get the wifi and mqtt connection to work first, then add more.

miloit, 2021-07-30:
@luebbe i build it from async wifi manager to async webserver to Matt client and this was running....now I try to remove the normal matt client and move on to async Mqtt.....

bertmelis, 2021-07-30:
Are you using SSL?

Because you set certificates for the WiFiClient but not for the MQTT client. Leftover from previous code?

miloit, 2021-07-30:
Yes, I am using TLS / SSL. I thought I activated it

mqttClient.setSecure(true);
  mqttClient.addServerFingerprint((const uint8_t*)mqttFingerprint);

closed 2021-08-21 by miloit