marvinroger/async-mqtt-client #260: Problem with Fingerprint verification

miloit, 2021-08-23:
Hello,
I have problem with accepting my fingerprint
I use a json to store my fingerprint and than converting it, but gettng the error about a bad fingerprint

```
{"mqtt_ssl":"69:66:84:1A:0E:E7:E5:51:30:82:60:A7:27:F2:A5:54:56:B9:15:19"}
void convert (char* from, char* to) {
    memcpy(to,"0x",2); to += 2;
    for (; *from; from++) {
        if (*from==':') {
            memcpy(to,", 0x",4); to += 4;
        } else {
            *to++ = *from;
        }
    }
    *to = 0;
}
mqttClient.addServerFingerprint((const uint8_t*)mqtt_sslconverted);

```
But if I use your solution everything is fine

`mqttClient.addServerFingerprint((const uint8_t[]){0x69, 0x66, 0x84, 0x1a, 0x0e, 0xe7, 0xe5, 0x51, 0x30, 0x82, 0x60, 0xa7, 0x27, 0xf2, 0xa5, 0x54, 0x56, 0xb9, 0x15, 0x19});
`

What is worng??

Pablo2048, 2021-08-23:
`(const uint8_t[]){....` mean binary array, your `convert()` converts text to another text... :roll_eyes:

closed 2021-08-23 by miloit

bertmelis, 2021-08-23:
You could parse the `string` to a valid `c array` with some logic and -for example- `https://en.cppreference.com/w/c/string/byte/strtol` or you can encode an array in your json and use it like this: https://arduinojson.org/v6/api/misc/copyarray/

Or use any other programming logic to your liking.