marvinroger/async-mqtt-client #261: Make topic const

bertmelis, 2021-09-03:
I'd like to propose a breaking change: make the topic in the onMessage callback `const`.

Under the hood the topic is indeed a `char*` but the end user shouldn't change the contents. I don't think there's a reason to anyway, but still...

https://github.com/marvinroger/async-mqtt-client/blob/89bf46485d5b60ce1e8e5e4d265a9c1570de3dc5/src/AsyncMqttClient/Callbacks.hpp#L15

Pablo2048, 2021-09-03:
+1 for this. I also suggest that the `payload` should be of type `const uint8_t *` instead of plain `char *` which is little confusing for newbies.

proddy, 2021-09-03:
+1. I typically add const everywhere. And yes to using the uint8_t instead of char. Maybe worth running the code through cppcheck/clangtidy or a service like codacy to see what other minor changes we could add?

bertmelis, 2021-09-03:
> +1 for this. I also suggest that the `payload` should be of type `const uint8_t *` instead of plain `char *` which is little confusing for newbies.

Yes, that would also be better indeed.