marvinroger/async-mqtt-client #262: Mqtt won't connect with network static IP

kbssa, 2021-09-14:
I am using WiFiManager configure my network connection, when using DHCP everything works like a charm, but when static IP is configured, the MQTT never connect, no matter what I do.

When I uncomment the line below in my code the Mqtt never connects, but everything else works (asyncWebserver, tcpServer).

`ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask);`

So the question is, Is that library compatible with static IP or I missed something ?

Pablo2048, 2021-09-14:
The library uses lwip ip stack, so there has to be problem somewhere else. Despite near zero informations which you have provided my crystal magic ball says that you want to connect to symbolic broker (eg. test.mosquitto.org) instead of direct ip address and your static configuration has missing the DNS server.

kbssa, 2021-09-14:
Sorry about providing poor informations.

When I create the WiFiManager object with the dnsServer is not enough ?

The snipet below is all the WiFiManager config I have in the code.

`
ESPAsync_WiFiManager ESPAsync_wifiManager(&server, &dnsServer, "DanfV3");
ESPAsync_wifiManager.setSaveConfigCallback(_saveConfigCallback);
ESPAsync_wifiManager.setConfigPortalChannel(0);
ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask);
ESPAsync_wifiManager.setBreakAfterConfig(true);
ESPAsync_wifiManager.setConfigPortalTimeout(120);
ESPAsync_wifiManager.setConnectTimeout(10);
ESPAsync_wifiManager.autoConnect("DanfV3");
`

kbssa, 2021-09-14:
> The library uses lwip ip stack, so there has to be problem somewhere else. Despite near zero informations which you have provided my crystal magic ball says that you want to connect to symbolic [...]

Your crystal magic ball were right !!
I changed the hostname by the IP itself and it worked, but, now I'm stucked as I don't know how to solve my dns problem.

Any help would be appreciated.

Pablo2048, 2021-09-15:
I'm not familiar with ESPAsync_WiFiManager ( hope that it's this library  https://github.com/khoih-prog/ESPAsync_WiFiManager), but poking into the source code it seems like you have to check if you have enabled `USE_CONFIGURABLE_DNS` option and then, in `setSTAStaticIPConfig()` you have to specify primary and secondary DNS server. You can use for example public Google DNS 8.8.8.8 as the primary one and your router ip as the secondary one.

kbssa, 2021-09-16:
> I'm not familiar with ESPAsync_WiFiManager ( hope that it's this library https://github.com/khoih-prog/ESPAsync_WiFiManager), but poking into the source code it seems like you have to check if you [...]

Thank you very much !!

It worked !!

Pablo2048, 2021-09-16:
Glad to hear that :-) . So please close this issue...

closed 2021-09-16 by kbssa