marvinroger/async-mqtt-client #264: TCP Disconnected

CurseStaff, 2021-10-05:
Hello Thank you for this great library,

[image]

it would seem that like several other person from the issues, I am faced with a connect / disconnect loop problem, anytime the system manages to authenticate itself to the mqtt server, and has published,

when the event on connected is triggered, we use ticker with a delay of 5 seconds to execute a void function which publishes a / out topic "hello" which succeeds 1/2 and after a subscibe of a topic / in , only when one of the two events (publish / subscibe) is triggered the error: Disconnected from MQTT, reason: TCP Disconnected. appears, and then it's a loop connect / disconnect error, however during the first initiation the publish works, and this hellish error loop only starts when a message is received.

bertmelis, 2021-10-06:
Do you have some code to share?

CurseStaff, 2021-10-06:
Main file:

//mqtt initialisation
  String mqtt_int_str = "CL_" + CORE_SERIAL + "/in";
  String mqtt_out_str = "CL_" + CORE_SERIAL + "/out";
  mqtt_int = mqtt_int_str.c_str();
  mqtt_out = mqtt_out_str.c_str();

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  
  mqttClient.onMessage(onMqttMessage);
  mqttClient.setServer(mqtt_host, mqtt_port);
  mqttClient.setCredentials(mqtt_user, mqtt_pass);
  connectToMqtt();
  delay(750);
  subscribemqtt();
  //debug("send hello to CurseLighting main server");
  //mqttClient.publish(mqtt_out, mqtt_out_qos, false, "HELLO");
  
  
  another included file:
  
  //mqtt section
AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;
Ticker mqttsubscribeTimer;

const char* mqtt_host = "mqtt.scintillement.fr";
uint16_t  mqtt_port = 1883;
const char* mqtt_user = "MQTT1";
const char* mqtt_pass = "836k4s260l4w5x746zrtvbk";

uint8_t  mqtt_int_qos = 0;
uint8_t  mqtt_out_qos = 0;

uint8_t qossub = 0;

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
}

void subscribemqtt() {
  debug("listen for CurseLighting main server");
  mqttClient.subscribe(mqtt_int, mqtt_int_qos);
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.print("Disconnected from MQTT, reason: ");
  if (reason == AsyncMqttClientDisconnectReason::TLS_BAD_FINGERPRINT) {
    Serial.println("Bad server fingerprint.");
  } else if (reason == AsyncMqttClientDisconnectReason::TCP_DISCONNECTED) {
    Serial.println("TCP Disconnected.");
  } else if (reason == AsyncMqttClientDisconnectReason::MQTT_UNACCEPTABLE_PROTOCOL_VERSION) {
    Serial.println("Bad server fingerprint.");
  } else if (reason == AsyncMqttClientDisconnectReason::MQTT_IDENTIFIER_REJECTED) {
    Serial.println("MQTT Identifier rejected.");
  } else if (reason == AsyncMqttClientDisconnectReason::MQTT_SERVER_UNAVAILABLE) {
    Serial.println("MQTT server unavailable.");
  } else if (reason == AsyncMqttClientDisconnectReason::MQTT_MALFORMED_CREDENTIALS) {
    Serial.println("MQTT malformed credentials.");
  } else if (reason == AsyncMqttClientDisconnectReason::MQTT_NOT_AUTHORIZED) {
    Serial.println("MQTT not authorized.");
  } else if (reason == AsyncMqttClientDisconnectReason::ESP8266_NOT_ENOUGH_SPACE) {
    Serial.println("Not enough space on esp8266.");
  }
  Serial.println(static_cast<uint8_t>(reason));
  //if (WiFi.isConnected()) {
    mqttReconnectTimer.once(5, connectToMqtt);
    mqttsubscribeTimer.once(8, subscribemqtt);
  //}
}

void onMqttMessage(char* topic, char* payload_in, AsyncMqttClientMessageProperties properties, size_t length, size_t index, size_t total) {
  uint8_t * payload = (uint8_t *) malloc(length + 1);
  memcpy(payload, payload_in, length);

  //change mode
  if (payload[0] == 'M') {
    
    mqttClient.publish(mqtt_out, mqtt_out_qos, false, "OK");
  }

  //change speed
  if (payload[0] == 'S') {
    Serial.println("command for update SPEED detected !");
    int value = (int) strtol((const char *) &payload[1], NULL, 10);
    value = value * 255;
    value = value / 65535;
    Serial.println(value);
    ws2812fx->setSpeed(value);
    
    mqttClient.publish(mqtt_out, mqtt_out_qos, false, "OK");
  }
  
  Serial.println("Publish received.");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  message: ");
  Serial.printf("%s\n", payload);

}

bertmelis, 2021-10-06:
Could you try with direct usage of `c_str()` instead of via `  mqtt_int = mqtt_int_str.c_str();
  mqtt_out = mqtt_out_str.c_str();`

So for example replace
```
mqttClient.subscribe(mqtt_int, mqtt_int_qos);
```
with
```
mqttClient.subscribe(mqtt_int_str.c_str(), mqtt_int_qos);
```

bertmelis, 2021-10-06:
Now, this is an asynchronous library which means that it is working based on events. So the "synchronous logic" doesn't (always) work:
- connect
- delay: wait for connection
- subscribe
- ...

Instead you have to do things when an event happens:
- connect
--> onConnect event: subscribe
--> onSubscribe: print log
--> onDisconnect event: reconnect
--> onMessage: handle message

So you should not use delay (and cannot in the event handlers). If delay is really needed, you set a flag in the handler and take care of the action on the Arduino `loop()`.

In your case there's at least `void subscribemqtt()` that needs to be moved to `void onMqttConnect(bool sessionPresent)`. You can also get rid of the subscribeTimer then.

CurseStaff, 2021-10-06:
Good evening,
your indications have been greatly helped and the problem no longer occurs, thank you for your quick and precise help!
good continuation !

closed 2021-10-06 by CurseStaff