marvinroger/async-mqtt-client #265: esp-8266 with SSL

KLelong, 2021-10-15:
Hi,
Has anyone success with building the library for esp-8266 in platformio, preferably on a platform version 3 or later ?
When I compile with `platform = espressif8266@3.2.0` I get : 
```
In file included from .pio/libdeps/d1_mini/ESPAsyncTCP/src/tcp_axtls.c:35:
.pio/libdeps/d1_mini/ESPAsyncTCP/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory
   44 | #include "include/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/d1_mini/lib774/ESPAsyncTCP/tcp_axtls.c.o] Error 1
In file included from .pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp:83:
.pio/libdeps/d1_mini/ESPAsyncTCP/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory
   44 | #include "include/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/d1_mini/lib774/ESPAsyncTCP/ESPAsyncTCP.cpp.o] Error 1
In file included from .pio/libdeps/d1_mini/AsyncMqttClient/src/AsyncMqttClient.hpp:22,
                 from .pio/libdeps/d1_mini/AsyncMqttClient/src/AsyncMqttClient.h:4,
                 from src/mqtt.h:1,
                 from src/main.cpp:24:
.pio/libdeps/d1_mini/ESPAsyncTCP/src/tcp_axtls.h:44:10: fatal error: include/ssl.h: No such file or directory
   44 | #include "include/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/d1_mini/src/main.cpp.o] Error 1
```
And when I set the platform pior to 3.0 (`platform = espressif8266@<3`) I get : 
```
.pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'err_t AsyncServer::_poll(tcp_pcb*)':
.pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp:1324:31: error: no matching function for call to 'AsyncClient::_recv(tcp_pcb*&, pbuf*&, int)'
         c->_recv(pcb, p->pb, 0);
                               ^
.pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp:1324:31: note: candidate is:
.pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp:565:6: note: void AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, err_t)
 void AsyncClient::_recv(std::shared_ptr<ACErrorTracker>& errorTracker, tcp_pcb* pcb, pbuf* pb, err_t err) {
      ^
.pio/libdeps/d1_mini/ESPAsyncTCP/src/ESPAsyncTCP.cpp:565:6: note:   candidate expects 4 arguments, 3 provided
Compiling .pio/build/d1_mini/lib546/ESP Async WebServer/AsyncWebSocket.cpp.o
*** [.pio/build/d1_mini/lib774/ESPAsyncTCP/ESPAsyncTCP.cpp.o] Error 1
```
Any suggestions or help ?
TIA
Koenraad

bertmelis, 2021-10-15:
I'll take a look soon. I'm upgrading a couple of my devices and one of them has SSL enabled...

bertmelis, 2021-12-01:
I guess the Arduino core has (finally) retired axTLS. The async TCP library has not moved to BearSSL.

This means that this library will not support TLS unless you're going to use Arduino core < v3. Latest is 2.7.4

closed 2021-12-01 by bertmelis

KLelong, 2021-12-02:
Thanks for investigating  this. 
This means async-MQTT-client can't be used for secured MQTT, because AFAIK async TCP can't be used with core < 3.
Maybe the README should be updated to state this ?

bertmelis, 2021-12-02:
It is possible but you'll need to use an updated fork.

In any case, it has to go into the docs indeed.

KLelong, 2021-12-02:
Can you point me to that fork ?

bertmelis, 2021-12-02:
I forgot, but you might want to check this: https://github.com/me-no-dev/ESPAsyncTCP/pull/129