marvinroger/async-mqtt-client #268: [HELP] what's the different between len and total

wwhai, 2021-11-17:
```
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
```

luebbe, 2021-11-17:
A big message may be split into several smaller parts, so:
len = The number of bytes of the current part of the message
index = The number of  the current part of the message
total = the number of bytes of the entire message

bertmelis, 2021-11-17:
A fictitious example how a big message works (Mind that this lib doesn't decide what "big" is. A message that is split up by the TCP layer is "big").

Client sends 3000 bytes:
first call
- len: 1400
- index: 0
- total: 3000
second call
- len: 1400
- index: 1400
- total 3000
third call
- len: 200
- index: 2800
- total: 3000

closed 2021-11-17 by bertmelis