marvinroger/async-mqtt-client #271: Replace async with WiFiClient?

bertmelis, 2021-12-02:
I don't know if it's doable but I'd like to check whether it is possible to replace the async tcp libs with the regular WiFiClient builtin libs. Async has one big disadvantage: the libs aren't (well) maintained and meanwhile they lack TLS.

I'm scrolling through the WiFiClient code to find incompatibilities but I have found none so far.

If somebody has some spare time... An extra set of eyes before I start this adventure is welcome!

luebbe, 2021-12-02:
I'll give it a spin in my homie based projects. There are some that I can easily update and monitor.

bertmelis, 2021-12-02:
Thing is, I always thought WiFiClient was blocking on connect and send. But I actually don't see the blocking code yet.

The main difference is that it doesn't work async with callbacks but it works non-blocking with polling. (That is: `mqttClient.loop()`in your Arduino `loop()`). For Home that isn't a problem because Homie has it's own loop-function.

I'd like to avoid this lib becoming obsolete.

proddy, 2021-12-02:
There is also https://github.com/yubox-node-org/AsyncTCPSock to consider, which is a re-write supporting TLS and backwards compatible. I haven't tried it yet but it's getting noticed in the AsyncWebServer world.

bertmelis, 2021-12-02:
> There is also https://github.com/yubox-node-org/AsyncTCPSock to consider, which is a re-write supporting TLS and backwards compatible. I haven't tried it yet but it's getting noticed in the AsyncWebServer world.

Only esp32 unfortunately

closed 2023-10-13 by bertmelis