marvinroger/async-mqtt-client #272: MQTT disconnections

AcuarioCat, 2021-12-07:
Hi, I'm running the sample ESP8266 code.
This is working ok publishing and subscribing but after every set of tests I'm getting an mqtt disconnect (TCP_DISCONNECTED) which then triggers another set of tests.

Is this the expected behaviour?

I want to use the library to listen for messages but I didn't expect to have to continually reconnect to the mqtt server.
I'm using the cloudmqtt.com public server.

bertmelis, 2021-12-07:
If the broker is disconnecting, it'd be very handy to have it's logs.
Is that possible?

AcuarioCat, 2021-12-07:
Here is the log with a couple of reconnects, hope this helps!
`2021-12-07 16:08:47: Client xbcfmbew already connected, closing old connection.
2021-12-07 16:08:47: Socket error on client xbcfmbew, disconnecting.
2021-12-07 16:08:47: New client connected from 185.13.76.175 as xbcfmbew (c1, k15, u'xbcfmbew').
2021-12-07 16:08:47: No will message specified.
2021-12-07 16:08:47: Sending CONNACK to xbcfmbew (0, 0)
2021-12-07 16:08:47: Received SUBSCRIBE from xbcfmbew
2021-12-07 16:08:47: 	test/lol (QoS 2)
2021-12-07 16:08:47: Sending SUBACK to xbcfmbew
2021-12-07 16:08:47: Sending PUBLISH to xbcfmbew (d0, q2, r1, m1, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Received PUBLISH from xbcfmbew (d0, q0, r1, m0, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Sending PUBLISH to xbcfmbew (d0, q0, r0, m0, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Received PUBLISH from xbcfmbew (d0, q1, r1, m22805, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Sending PUBACK to xbcfmbew (Mid: 22805)
2021-12-07 16:08:47: Sending PUBLISH to xbcfmbew (d0, q1, r0, m2, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Received PUBLISH from xbcfmbew (d0, q2, r1, m22806, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Sending PUBREC to xbcfmbew (Mid: 22806)
2021-12-07 16:08:47: Received PUBREL from xbcfmbew (Mid: 22806)
2021-12-07 16:08:47: Sending PUBCOMP to xbcfmbew (Mid: 22806)
2021-12-07 16:08:47: Sending PUBLISH to xbcfmbew (d0, q2, r0, m3, 'test/lol', ... (6 bytes))
2021-12-07 16:08:47: Received PUBREC from xbcfmbew (Mid: 1)
2021-12-07 16:08:47: Sending PUBREL to xbcfmbew (Mid: 1)
2021-12-07 16:08:48: Received PUBCOMP from xbcfmbew (Mid: 1)
2021-12-07 16:08:48: Received PUBACK from xbcfmbew (Mid: 2)
2021-12-07 16:08:48: Received PUBREC from xbcfmbew (Mid: 3)
2021-12-07 16:08:48: Sending PUBREL to xbcfmbew (Mid: 3)
2021-12-07 16:08:48: Received PUBCOMP from xbcfmbew (Mid: 3)
2021-12-07 16:08:49: New connection from 185.13.76.175 on port 19167.
2021-12-07 16:08:49: Client xbcfmbew already connected, closing old connection.
2021-12-07 16:08:49: Socket error on client xbcfmbew, disconnecting.
2021-12-07 16:08:51: New client connected from 185.13.76.175 as xbcfmbew (c1, k15, u'xbcfmbew').
2021-12-07 16:08:51: No will message specified.
2021-12-07 16:08:51: Sending CONNACK to xbcfmbew (0, 0)
2021-12-07 16:08:51: Received SUBSCRIBE from xbcfmbew
2021-12-07 16:08:51: 	test/lol (QoS 2)
2021-12-07 16:08:51: Sending SUBACK to xbcfmbew
2021-12-07 16:08:51: Sending PUBLISH to xbcfmbew (d0, q2, r1, m1, 'test/lol', ... (6 bytes))
2021-12-07 16:08:51: Received PUBLISH from xbcfmbew (d0, q0, r1, m0, 'test/lol', ... (6 bytes))
2021-12-07 16:08:51: Sending PUBLISH to xbcfmbew (d0, q0, r0, m0, 'test/lol', ... (6 bytes))
2021-12-07 16:08:51: Received PUBLISH from xbcfmbew (d0, q1, r1, m22808, 'test/lol', ... (6 bytes))
2021-12-07 16:08:51: Sending PUBACK to xbcfmbew (Mid: 22808)
2021-12-07 16:08:51: Sending PUBLISH to xbcfmbew (d0, q1, r0, m2, 'test/lol', ... (6 bytes))
2021-12-07 16:08:52: Received PUBLISH from xbcfmbew (d0, q2, r1, m22809, 'test/lol', ... (6 bytes))
2021-12-07 16:08:52: Sending PUBREC to xbcfmbew (Mid: 22809)
2021-12-07 16:08:52: Received PUBREL from xbcfmbew (Mid: 22809)
2021-12-07 16:08:52: Sending PUBCOMP to xbcfmbew (Mid: 22809)
2021-12-07 16:08:52: Sending PUBLISH to xbcfmbew (d0, q2, r0, m3, 'test/lol', ... (6 bytes))
2021-12-07 16:08:52: Received PUBREC from xbcfmbew (Mid: 1)
2021-12-07 16:08:52: Sending PUBREL to xbcfmbew (Mid: 1)
2021-12-07 16:08:52: Received PUBCOMP from xbcfmbew (Mid: 1)
2021-12-07 16:08:52: Received PUBACK from xbcfmbew (Mid: 2)
2021-12-07 16:08:52: Received PUBREC from xbcfmbew (Mid: 3)
2021-12-07 16:08:52: Sending PUBREL to xbcfmbew (Mid: 3)
2021-12-07 16:08:52: Received PUBCOMP from xbcfmbew (Mid: 3)
2021-12-07 16:08:54: New connection from 185.13.76.175 on port 19167.
2021-12-07 16:08:54: Client xbcfmbew already connected, closing old connection.
2021-12-07 16:08:54: Socket error on client xbcfmbew, disconnecting.
2021-12-07 16:08:54: New client connected from 185.13.76.175 as xbcfmbew (c1, k30, u'xbcfmbew').
2021-12-07 16:08:54: No will message specified.
2021-12-07 16:08:54: Sending CONNACK to xbcfmbew (0, 0)`

bertmelis, 2021-12-07:
So your client is trying to connect while it is already connected.

Do you use the example, unchanged?

AcuarioCat, 2021-12-07:
yes, just added my credentials

AcuarioCat, 2021-12-07:
Ah, maybe I have found something - using usernames rather than the default account seems to stop the disconnects..

bertmelis, 2021-12-07:
What exactly did you do? Added credentials? Same credentials to different devices?
I admit I didn't explore the MQTT spec wrt this topic.

ajinkya-open, 2022-12-28:
similar problem with non async library too, so luck yet