marvinroger/async-mqtt-client #273: Connect with wrong credentials appears like a valid connection

AcuarioCat, 2021-12-09:
When connecting using credentials that are incorrect it appears the connection is successful.
I'm using the default esp8266 sample.

This is the log from a valid connect:
Connecting to MQTT...
Connected to MQTT.
Session present: 0
Subscribing at QoS 2, packetId: 85
Publishing at QoS 0
Publishing at QoS 1, packetId: 86
Publishing at QoS 2, packetId: 87
Subscribe acknowledged.
  packetId: 85
  qos: 2
Publish received.
.
.
.

This is with invalid credentials:
Connecting to MQTT...
Connected to MQTT.
Session present: 0
Subscribing at QoS 2, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
Subscribe acknowledged.
  packetId: 1
  qos: 2
Publish acknowledged.
  packetId: 2
Publish acknowledged.
  packetId: 3

bertmelis, 2021-12-09:
That seems to be an issue with your broker. Why would it send acknowledgements when it should reject the client?

AcuarioCat, 2021-12-09:
It seems odd. I'm using CloudMQTT which is commercial so I would expect it to behave properly.