marvinroger/async-mqtt-client #276: subscribe is poorly documented

NohWayJose, 2022-01-05:
Please will you provide at least ONE well commented example of the use of `mqttClient.subscribe` all the way through to consuming a message on a subscribed topic and covering where the subscribe goes (is it in the `setup()` function?) and how the message arrives during the execution of the `loop()` and how to capture it?

NohWayJose, 2022-01-05:
For context, I use a button to publish a message to _start_  a fan for a few minutes. NodeRed subscribes and manages that. When the time is up, NodeRed publishes a _resume_ message, which the ESP8266 should subscribe to. Capturing that message will allow code on the ESP8266 to reset the mode to normal, where the fan is controlled by a humidity sensor.

The button is basically a timed override to get rid of cooking/bathroom smells where humidity is low, while the standard mode requires no user input, to clear steam/humidity.

NohWayJose, 2022-01-06:
Your FullyFeatured-ESP8266.ino is the nearest example I've found but 
```
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("Publish received.");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}
```

You have `Serial.println("Publish received.");` which is ambiguous to me is it a `subscribed message received` or a `message published`?

Also, there's nothing in the `loop()` showing how anything's used.

I think this is a great library. It's just that I am a UXer not a programmer and I need the examples to be complete and easy. I'm sure there will be others out there who'd appreciate that little extra too. Thanks in advance :-D

Pablo2048, 2022-01-06:
This is **Async** library, based on AsyncTCP so there is nothing in the loop() and given callbacks are called asynchronously. There are some limitations in this approach (no delay() in the callback, etc.) so please read also all documentation for the AsyncTCP library. The `onMqttMessage` callback is called on message receive (client has received publish from another source).