marvinroger/async-mqtt-client #280: AsyncMQTTClient::publish() is crashing the ESP32-CAM

jollyto, 2022-03-03:
The AsyncMQTTClient::publish() is crashing the ESP32-CAM

Here is the code:

void takePublishPhoto(){
  Serial.println("-- ====================");
   Serial.println("-- Taking a photo ...");
   
   camera_fb_t* fb = NULL;
   fb = esp_camera_fb_get();

  Serial.println("-- Photo taken!");
  
  if (!fb)
  {
    Serial.println("Camera capture failed !!!");
    return;
  }
    
  Serial.print("-- Frame Size : ");Serial.println(fb->len);Serial.println();
  Serial.println("-- Publishing ...");

  String encoded = base64::encode(fb->buf, fb->len );
  Serial.print("-- Encoded Size : ");Serial.println(encoded.length());Serial.println();
  
  printRamUsage("Before publish photo");
 
  int result = mqttClient.publish( PUB_TOPIC_PHOTO, 0, false, 
                                 (const char*)encoded.c_str(), encoded.length()); // fb->len );  
  esp_camera_fb_return(fb);
  
  }

Here is  the output :
09:31:16.137 -> -- ====================
09:31:16.137 -> -- Taking a photo ...
09:31:16.173 -> -- Photo taken!
09:31:16.173 -> -- Frame Size : 23944
09:31:16.209 -> 
09:31:16.209 -> -- Publishing ...
09:31:16.209 -> -- Encoded Size : 31928
09:31:16.209 -> 
09:31:16.209 -> 
09:31:16.209 -> -- RAM USAGE Before publish photo --
09:31:16.209 -> Total heap : 323968
09:31:16.209 -> Free heap  : 104964
09:31:16.209 -> PSRAM found: Yes
09:31:16.209 -> Total PSRAM: 4194252
09:31:16.209 -> Free PSRAM : 4194252
09:31:16.209 -> Used PSRAM : 0
09:31:16.209 -> Free Memory: 40864
09:31:16.209 -> 
09:31:16.209 -> abort() was called at PC 0x4014b1e7 on core 1

Any ideas what may crash when calling the publish() method.

luebbe, 2022-03-04:
@jollyto since you put that much effort in your reply, you are surely expecting everyone here to grab their crystal balls and debug your code for you...
/sarcasm off

closed 2022-03-04 by jollyto

luebbe, 2022-03-04:
Funny thing is: people here are willing to help you. You just have to provide the information that allows us to help

jollyto, 2022-03-04:
And I really appreciated. But I need to get prepared to provide more detail (stack trace, ect). That s why I closed the issue. As soon as I get organized I’ll reopen the issue.  
My sincerely apologies if I offended the group with my short explanation of the problem.  

— José