marvinroger/async-mqtt-client #281: OnDisconnect Never Called when Internet is Off.

rndalteem, 2022-03-07:
I have implemented my code using Lib and while testing i found when internet is turned off library never call OnDisconnect Callback. I have attached log when Internet is off.
I am using Latest Release 0.9.0

[image]

bertmelis, 2022-04-20:
To have things clear: keepalive is kept default and the connection to the broker is interrupted.

Yes?

proddy, 2022-04-20:
And clean session is disabled?

ktannady, 2022-04-21:
Hello, thanks for your reply.

> To have things clear: keepalive is kept default and the connection to the broker is interrupted.

> And clean session is disabled?

Yes, the connection to the broker is interrupted, I have never called the keepalive and clean session function so I think it is the default 15 seconds for keepalive and true for the clean session.

for some quick fix, I added the `ESP8266Ping` function to check if there is a connection to the broker, if not I just call the `mqttClient.disconnect()` function to trigger the `onDisconnect()`.

bertmelis, 2022-04-21:
I have to add some extra debug statements at the keepalive handling. It should disconnect at twice the keepalive.

ktannady, 2022-04-22:
> 

Hello, thanks for your reply.
I don't know how and where to compile the library with debug statement, can you help me?

bertmelis, 2022-04-23:
I'm on holiday now and only limited access to my computer. Are you using Arduino IDE or Platformio or a different IDE?

ktannady, 2022-04-23:
> I'm on holiday now and only limited access to my computer. Are you using Arduino IDE or Platformio or a different IDE?

I'm using Platformio.

iggy80, 2022-04-29:
The problem is located in the _handleQueue method (fileAsyncMqttClient.cpp line 394).
The _lastPingRequestTime variable is reset to 0 at line 410, then the _onPoll keeps sending PingRequests.
Just remove or comment the _lastPingRequestTime = 0; and the disconnect now fires after keep alive delay when broker is disconnected.....

bertmelis, 2022-04-29:
@iggy80 PRs welcome!

rndalteem, 2022-05-04:
@iggy80 I have tested its working now.Thanks.

closed 2022-05-04 by rndalteem

ktannady, 2022-05-13:
Hello, there is no PR yet?

bertmelis, 2022-05-13:
> Hello, there is no PR yet?

#284