marvinroger/async-mqtt-client #285: Unknown reason for failed Publish packet.

ambarusa, 2022-05-06:
Hello guys. Currently my issue is that not every packet is failed to be published. The same function is used for every publish.
[image]

Do you know a way how to debug why is the publishing service returning 0?

ambarusa, 2022-05-06:
Its Platformio.

Here the publish works:
`void onMqttConnect(bool sessionPresent)
{
   mqtt_reconn_ticker.stop();
   Serial.println("MQTT: Connected!");

   amqtt_client.subscribe(mqtt_sw_cmd_topic, qossub);

   amqtt_client.publish(mqtt_availability_topic, qospub, true, mqtt_birth_payload);
   Mqtt_state_publish();
   Mqtt_discovery_publish();
}`

If I get a message in the `.../set` topic, pub state publish is failing
`void onMqttMessage(char *topic, char *payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
{
   payload[len] = '\0';
   Serial.printf("\nMQTT: Recieved [%s]: %s\n", topic, payload);

   if (strcmp(payload, "ON") == 0)
   {
      Set_power_relay(true);
   }
   else if (strcmp(payload, "OFF") == 0)
   {
      Set_power_relay(false);
   }
   Mqtt_state_publish();
}
`
`void Mqtt_state_publish()
{
   const char* payload = (Get_power_relay()) ? "ON" : "OFF";
   uint16_t id = amqtt_client.publish(mqtt_sw_topic, qospub, true, payload);
   Serial.printf("MQTT State Send [ID %i]: [%s]: %s\n", id, mqtt_sw_topic, payload);
}`

Everything is read out correctly, my only issue is the failed publish operation.

bertmelis, 2022-05-06:
Add the following build flags to your platformio.ini:
```
build_flags =
  -D DEBUG_ESP_PORT=Serial
  -D DEBUG_ASYNC_MQTT_CLIENT
```

It'll give a bit more info about what's happening.

ambarusa, 2022-05-06:
Thanks! It gets me somewhere.
```
MQTT: Recieved [gaggia_smart/switch/set]: OFF
Hardware: Set Power Relay to OFF
PUBLISH
new back #3
snd #3: 26/26
p #3 rel
MQTT State Send [ID 1]: [gaggia_smart/switch]: OFF
No poison after block at: 0x3fff13b0, actual data: 0x0 0xa5 0xa5 0xa5

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: sys
sp: 3fffec60 end: 3fffffb0 offset: 0000
3fffec60:  00000000 3ffef1e4 00000001 3ffe89d2
3fffec70:  000000fe 00000000 00000000 00000000
```

bertmelis, 2022-05-06:
Now you'll have to decode your stack trace...

ambarusa, 2022-05-06:
You are absolutely right!! My problem there was that the received payload looked like: `ON������␂(␂���␂(␂`. I wanted to use '\0' to cleanup the payload.

ambarusa, 2022-05-06:
I will also go with `String payload = ((String)payload_raw).substring(0,len);` from [this](https://github.com/marvinroger/async-mqtt-client/issues/104#issuecomment-388981994) conversation, I recently discovered. It will be easier to process the data as well with strings. Thank you very much.

closed 2022-05-06 by ambarusa