marvinroger/async-mqtt-client #296: large mqtt-publish messages (ESP32-CAM board for jpeg pictures)

GnomiBerlin, 2023-05-21:
Hello,
by using this async-mqtt-client or the extended work of @khoih-prog  https://github.com/khoih-prog/AsyncMQTT_ESP32 (seems to be stopped) I have  some questions.
Using synchronous mqtt  with pubsubclient, I could publish >200 kB jpeg pictures via mqtt. In moving to asynchronous mqtt,  there seems to be a limit at around 115kB with async-mqtt-client (the ESP reboots)

Questions:
@bertmelis development branch enhancements were merged in 0.9.0 latest release?
There is still a size limit for publish mqtt messages here?
Is there any other  more supported async-mqtt solution without the limits?

Thanks Joerg

bertmelis, 2023-05-21:
I haven't implemented large payloads in this library.

I built another one from zero without size restrictions: https://github.com/bertmelis/espMqttClient

GnomiBerlin, 2023-05-22:
thanks @bertmelis for the fast response.

I adapted all and inserted your library into the IDE to test this new solution but receive the following **errors** during compilation of your code:

```
In file included from D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/MqttClient.h:22:0,
                 from D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/MqttClientSetup.h:14,
                 from D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/espMqttClientAsync.h:18,
                 from D:\SYS_ext\Joerg\Documents\Arduino\ESP32-CAM_TESTallasync\ESP32-CAM_TESTallasync.ino:16:
D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/Packets/Packet.h: In constructor 'espMqttClientInternals::Packet::Packet(espMqttClientTypes::Error&, uint16_t, const char*, uint8_t, const char*, uint8_t, Args&& ...)':
D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/Packets/Packet.h:100:43: error: expected ',' before ')' token
     static_assert(sizeof...(Args) % 2 == 0);
                                           ^
D:\SYS_ext\Joerg\Documents\Arduino\libraries\espMqttClient-1.4.2\src/Packets/Packet.h:100:43: error: expected string-literal before ')' token

```

**And two questions if possible:**
if using this:

```
void onMqttMessage(const espMqttClientTypes::MessageProperties& properties, const char* topic, const uint8_t* payload, size_t len, size_t index, size_t total)
{
  (void) payload;

  String payloadTemp;
  for (int i = 0; i < len; i++) {
    payloadTemp += (char)payload[i];
  }

```
a)  what is the intention of "(void) payload;"  here?
b) to get a string from the new payload format, is this the correct way?

Thanks Joerg

closed 2023-05-22 by bertmelis