marvinroger/async-mqtt-client #297: Delay when receiving message

s0301132, 2023-06-05:
Hi @bertmelis, I have 50+ esp32 connected to MQTT using this library, however sometimes it have huge delay (2 - 5 sec, sometimes over 10 sec) between publish and onMessage. I tried some configuration such as longer keep alive time, both subscribe and publish running qos 0 etc but the issue remain. Do you know how to improve the delay around 1 - 2 sec?

As you mention it may be `async` library problem:

> The main difference is that it doesn't work async with callbacks but it works non-blocking with polling. (That is: mqttClient.loop()in your Arduino loop()). For Home that isn't a problem because Homie has it's own loop-function. This means the lib will be a bit slower because messages will be arriving on the next entry of the loop() function instead of immediately in a callback from LWIP. I guess we're talking milliseconds.

If I empty the `mqttclient.loop()`, can it reduce the dalay time? Thanks

bertmelis, 2023-06-05:
wrong repo?

s0301132, 2023-06-05:
Oh, sorry about that @bertmelis, just realised it, I'll try your repo without async one, thanks a lot.

closed 2023-06-05 by s0301132