marvinroger/async-mqtt-client #305: Disconnected from MQTT. Reason: TCP_DISCONNECTED

lbarcella, 2024-06-07:
Hi,
  i use PlatformIO with:
platform = espressif32 
board = esp32dev

My MQTT server is Mosquitto based
With framework-arduinoespressif32 @ 2.17 all work fine.

The arduino framework is now updated to framework-arduinoespressif32 @ 3.0.0
I have update the wifi events.
In a first time all works fine, but next few minutes the connections to MQTT become very instable

I receive a lot of events TCP_DISCONNECTED and next TCP_CONNECTED

I have test the same code with broker.hivemq.com free version and all works fine !

Any idea?
Thank you
L.

jeremyvnc, 2024-07-02:
I am running into the same issue. Have you found any workarounds?

lbarcella, 2024-07-03:
Hi, 
  yes, by my side i have add the clientid on the client call and all work fine.