marvinroger/async-mqtt-client #308: compilation error 'class MDNSResponder' has no member named 'IP'

crbyxwpzfl, 2024-08-19:
Hi I just tried to compile your `examples/FullyFeatured-ESP32/FullyFeatured-ESP32.ino` and got three errors.
This one I was not able to resolve
```c
/Users/***/Documents/Arduino/libraries/AsyncMqttClient/src/AsyncMqttClient.cpp:142:28: error: 'class MDNSResponder' has no member named 'IP'
  142 |                 _ip = MDNS.IP(0);
      |                            ^~
```

<details>
<summary>These to I resolved as the error said.</summary>

```c
/Users/***/Desktop/test/test.ino: In function 'void WiFiEvent(arduino_event_id_t)':
/Users/***/Desktop/test/test.ino:35:14: error: 'SYSTEM_EVENT_STA_GOT_IP' was not declared in this scope; did you mean 'IP_EVENT_STA_GOT_IP'?
   35 |         case SYSTEM_EVENT_STA_GOT_IP:
      |              ^~~~~~~~~~~~~~~~~~~~~~~
      |              IP_EVENT_STA_GOT_IP
/Users/***/Desktop/test/test.ino:41:14: error: 'SYSTEM_EVENT_STA_DISCONNECTED' was not declared in this scope; did you mean 'WIFI_EVENT_STA_DISCONNECTED'?
   41 |         case SYSTEM_EVENT_STA_DISCONNECTED:
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              WIFI_EVENT_STA_DISCONNECTED
```
I replaced [line 35](https://github.com/marvinroger/async-mqtt-client/blob/58cc70449fbeb0ffeb5740da91870de089a8a21d/examples/FullyFeatured-ESP32/FullyFeatured-ESP32.ino#L35) with `case IP_EVENT_STA_GOT_IP:` 
and replaced [line 41](https://github.com/marvinroger/async-mqtt-client/blob/58cc70449fbeb0ffeb5740da91870de089a8a21d/examples/FullyFeatured-ESP32/FullyFeatured-ESP32.ino#L41) with `case WIFI_EVENT_STA_DISCONNECTED:` as the compiler error said.
</details>

I compiled for an Adafruit Feather Esp32s3.
Please close this whenever. This likely is just me failing to read some docs correctly as no one reported this. Anyway Perhaps someone has a tip for me. And Thank you for this library!

doko-rr, 2024-08-20:
I encountered the same error as well.

crbyxwpzfl, 2024-08-20:
The second two issues perhaps are form Arduino core v3 see [here](https://forum.arduino.cc/t/wifi-connection-stopped-after-updates/1265257).
[theelims/PsychicMqttClient](https://github.com/theelims/PsychicMqttClient) lib is v3 compatible since this [release](https://github.com/theelims/PsychicMqttClient/releases/tag/0.2.0).
Sorry for suggesting an other lib here but I dont know enough to pr a fix.

knopserl, 2024-08-23:
I have the same compile error since I switched from AsyncMQTT_ESP32 to this library and it seem not to work with the new ESP32 API's or ???

rustysoma, 2024-08-26:
In the file AsyncMqttClient.cpp on line 142, replace `_ip = MDNS.IP(0);` with `_ip = MDNS.address(0);`

crbyxwpzfl, 2024-08-26:
thank you @rustysoma this solves it. ill close this now.

closed 2024-08-26 by crbyxwpzfl

RuiSantosdotme, 2024-08-28:
@luebbe @cturqueti  can you make this change to the develop branch? 

Otherwise the current version is not compiling with ESP32 Arduino core v3

In the file AsyncMqttClient.cpp on line 142, replace _ip = MDNS.IP(0); with _ip = MDNS.address(0);

luebbe, 2024-08-29:
Can you please provide a PR?